Livox-SDK/Livox-SDK #178: 如何像Livox Viewer那样将Reflectivity值映射为color值

RangeLiu, 2022-06-12:
Hi，如题，怎样像livox viewer那样将Reflectivity值映射为伪色彩值呢？有没有Livox Viewer那样的伪色彩表提供？谢谢！

Livox-SDK, 2022-06-12:
Hi RangeLiu,

可将Livox点云数据格式中反射率这一位解析出来，对不同数值区间的反射率值进行颜色表征。
反射率数值所在位置可以查看：[Livox SDK Communication Protocol](https://github.com/Livox-SDK/Livox-SDK/wiki/Livox-SDK-Communication-Protocol-Cn)

RangeLiu, 2022-06-12:
> Hi RangeLiu,
> 
> 可将Livox点云数据格式中反射率这一位解析出来，对不同数值区间的反射率值进行颜色表征。 反射率数值所在位置可以查看：[Livox SDK Communication Protocol](https://github.com/Livox-SDK/Livox-SDK/wiki/Livox-SDK-Communication-Protocol-Cn)

谢谢回复。我目前也是这么做的，对于反射率和颜色之间的映射，我参考了https://github.com/Livox-SDK/Livox-SDK/wiki/Livox-Viewer#1-color-coding-strategy。
但并不能得到Livox Viewer中这样的颜色效果，请问Livox Viewer中右下角的这个颜色映射表我可以从哪里找到？谢谢！
[image]

RangeLiu, 2022-06-12:
我翻阅了一下之前的issue，#135应该是跟我相同的问题，谢谢！

RangeLiu, 2022-06-13:
已经明白怎样处理的了，请关闭这个问题，谢谢。

85256638, 2024-06-20:
> 已经明白怎样处理的了，请关闭这个问题，谢谢。

如果知道怎么解决了，请贴出来解决方法，这个是常识啊，帮助后面的人。谢谢！