Livox-SDK/Livox-SDK #198: in ubuntu 20.04 ,static IP can't set successfully!

PengKunPROO, 2023-03-18:
I use avia in ubuntu 20.04 ,and set ip to 192.168.1.50 with command like this below
sudo ifconfig enp0s1 192.168.1.50
and the network between avia and computer seems blocked!
[image]
the command above is not working,I don't know why the avia  connected to my computer failed!

ArcticSnow, 2024-02-17:
I am running in a similar issue, where I can ping the lidar, but cannot connect to it in any ways. Have you solved the issue @PengKunPROO ?

C4Consulting, 2024-10-19:
Wondering if either of you found a solution to this?
I'm likely just going to replace my AVIA with an ultra wide or hemispherical sensor that works on 20.04 and above
New Jetson devices aren't supporting before that SW version anyway and that's the base of my design.

ArcticSnow, 2024-10-23:
Hello @C4Consulting , if I recall well, you need to turn dow all firewall on the network of the lidar, and if the lidar does not connect, try a hard reboot of the lidar. I have not run in this issue anymore using the Avia.

C4Consulting, 2024-10-23:
Thanks for the response! I was using one robosense b pearl and one avía in
my project but since I couldn’t figure it out I just sold the avía for
another b pearl. It was my preference anyway but was initially cost
prohibitive until I found a deal on a used one.

These guys are truly abandoning their best product with this lack of avia
compatibility/support

Cheers

*Christopher Allen*
*Director of Virtual, Design, & Construction*

* <http://consultc4.com>          *
*O:301.254.5556*
*E: ***@***.*** ***@***.***>*

*  <http://instagram.com/consultc4>
<https://www.linkedin.com/company/consultc4/>*
*Creative Coordinated Consulting Company LLC*