Livox-SDK/Livox-SDK #202: Livox Avia GPS Synchronization

LiShuaixin, 2023-04-27:
Hello! I'd like to synchronize the Avia lidar with GPRMC, and I connected wires following the wiki tutorial. Everything looks fine. Before launching it in ROS, I tested the synchronization status. 

1. Using DSK
I ran lidar_utc_sync in sdk samples `./lidar_utc_sync` and it outputted:
```
Livox SDK version 2.3.0
Local broadcast code : 000000000000001
Invalid local broadcast code : 000000000000001
No broadcast code was added to whitelist, swith to automatic connection mode!
Livox-SDK init success!
Init lds lidar success!
Start discovering device.
[Baudrate]: 2 13
Set baudrate success.
Synchro start successRmc: $GPRMC,011205.20,A,3951.00585,N,11617.17281,E,0.056,,270423,,,A*7D
[2023-04-27 09:12:05.827] [console] [info]  Broadcast broadcast code: 3JEDK620010U451  [device_discovery.cpp] [OnBroadcast] [150]
In automatic connection mode, will connect 3JEDK620010U451
[2023-04-27 09:12:05.828] [console] [info] LocalIP: 192.168.1.2  [device_discovery.cpp] [OnBroadcast] [201]
[2023-04-27 09:12:05.828] [console] [info] DeviceIP: 192.168.1.145  [device_discovery.cpp] [OnBroadcast] [202]
[2023-04-27 09:12:05.828] [console] [info] Command Port: 55501  [device_discovery.cpp] [OnBroadcast] [203]
[2023-04-27 09:12:05.828] [console] [info] Data Port: 56001  [device_discovery.cpp] [OnBroadcast] [204]
[2023-04-27 09:12:05.831] [console] [info] New Device  [device_discovery.cpp] [OnData] [99]
[2023-04-27 09:12:05.831] [console] [info] Handle: 0  [device_discovery.cpp] [OnData] [100]
[2023-04-27 09:12:05.831] [console] [info] Broadcast Code: 3JEDK620010U451  [device_discovery.cpp] [OnData] [101]
[2023-04-27 09:12:05.831] [console] [info] Type: 7  [device_discovery.cpp] [OnData] [102]
[2023-04-27 09:12:05.831] [console] [info] IP: 192.168.1.145  [device_discovery.cpp] [OnData] [103]
[2023-04-27 09:12:05.831] [console] [info] Command Port: 55501  [device_discovery.cpp] [OnData] [104]
[2023-04-27 09:12:05.831] [console] [info] Data Port: 56001  [device_discovery.cpp] [OnData] [105]
[2023-04-27 09:12:05.832] [console] [info]  Send Command: Set 0 Id 2 Seq 3  [command_channel.cpp] [Send] [243]
[2023-04-27 09:12:05.835] [console] [info]  Recieve Ack: Set 0 Id 2 Seq 3  [command_handler.cpp] [OnCommand] [118]
[WARNING] Lidar sn: [3JEDK620010U451] Connect!!!
Device Working State 5
Device State Error Code 0X00000000
Device feature 0
[2023-04-27 09:12:05.835] [console] [info]  Send Command: Set 0 Id 2 Seq 4  [command_channel.cpp] [Send] [243]
[2023-04-27 09:12:05.837] [console] [info]  Recieve Ack: Set 0 Id 2 Seq 4  [command_handler.cpp] [OnCommand] [118]
firm ver: 11.8.0.6
[2023-04-27 09:12:05.891] [console] [info]  Update State to 1, device connect true  [device_manager.cpp] [UpdateDeviceState] [281]
[WARNING] Lidar sn: [3JEDK620010U451] StateChange!!!
Device Working State 1
Device State Error Code 0X0000c200
Device feature 0
Rmc: $GPRMC,011206.20,A,3951.00587,N,11617.17282,E,0.063,,270423,,,A*79
[2023-04-27 09:12:06.554] [console] [info]  Send Command: Set 1 Id 10 Seq 6  [command_channel.cpp] [Send] [243]
[2023-04-27 09:12:06.557] [console] [info]  Recieve Ack: Set 1 Id 10 Seq 6  [command_handler.cpp] [OnCommand] [118]
OnLidarSetRmcSyncTimeCallback statue 0 handle 0 response 0. 
Rmc: $GPRMC,011207.20,A,3951.00592,N,11617.17282,E,0.015,,270423,,,A*7D
[2023-04-27 09:12:07.546] [console] [info]  Send Command: Set 1 Id 10 Seq 9  [command_channel.cpp] [Send] [243]
[2023-04-27 09:12:07.549] [console] [info]  Recieve Ack: Set 1 Id 10 Seq 9  [command_handler.cpp] [OnCommand] [118]
OnLidarSetRmcSyncTimeCallback statue 0 handle 0 response 0. 
[WARNING] Lidar sn: [3JEDK620010U451] StateChange!!!
Device Working State 1
Device State Error Code 0X00008200
Device feature 0
Rmc: $GPRMC,011208.20,A,3951.00598,N,11617.17282,E,0.056,,270423,,,A*7F
[2023-04-27 09:12:08.553] [console] [info]  Send Command: Set 1 Id 10 Seq 11  [command_channel.cpp] [Send] [243]
[2023-04-27 09:12:08.557] [console] [info]  Recieve Ack: Set 1 Id 10 Seq 11  [command_handler.cpp] [OnCommand] [118]
OnLidarSetRmcSyncTimeCallback statue 0 handle 0 response 0. 
Rmc: $GPRMC,011209.20,A,3951.00605,N,11617.17280,E,0.078,,270423,,,A*77
[2023-04-27 09:12:09.545] [console] [info]  Send Command: Set 1 Id 10 Seq 13  [command_channel.cpp] [Send] [243]
[2023-04-27 09:12:09.549] [console] [info]  Recieve Ack: Set 1 Id 10 Seq 13  [command_handler.cpp] [OnCommand] [118]
OnLidarSetRmcSyncTimeCallback statue 0 handle 0 response 0.
```
It shows that the LiDAR is connected and GPRMC is parsed correctly. But some errors is threw out:
```
Device State Error Code 0X00000000
Device feature 0
...
...
Device Working State 1
Device State Error Code 0X0000c200
Device feature 0
...
...
Device Working State 1
Device State Error Code 0X00008200
Device feature 0
```

2. Using Livox Viewer
Then I used Livox Viewer to check the status and I showed:
[image]

I seems like some errors occurs.

4. Checking Data 
Finally, I recorded a data sample and checked the timestamp type
[image]

I seems like the LiDAR is synchronized.

Now I'm confused whether the LiDAR is synchronized. Please help me to check the issue, many thanks!

Yneng, 2023-05-18:
Hello, I would like to ask how your GPS device is connected to the Livox Converter 2.0 interface. We use D-RTK and do not know how to connect the hardware

supercrazysam, 2023-10-31:
> Hello! I'd like to synchronize the Avia lidar with GPRMC, and I connected wires following the wiki tutorial. Everything looks fine. Before launching it in ROS, I tested the synchronization [...]

Hi @LiShuaixin did you figured out what was the issue? Just curious

TokyoWarfare, 2025-11-14:
> Hello, I would like to ask how your GPS device is connected to the Livox Converter 2.0 interface. We use D-RTK and do not know how to connect the hardware

Hi. I'm as confused as you, could you share some details on wetehr livox Converer 2.0 needs to receive PPS form the GPS or sendint the PPS +Time tia TTL-USB converter is enough for the SDK to sync the sensor..