Livox-SDK/Livox-SDK #212: PPS + NMEA (GPS_sync) supplied, but stuck in timestamp_type 4 (PPS)

supercrazysam, 2023-10-31:
Hi,   

I am supplying PPS signal and sending GPRMC to the "enable_timesync" port,  however the timestamp_type always stuck at 4 (PPS).  

It never gets to 3 (GPS).   

Also, in the livox viewer, it says sync abnormal but does not offer more information other than that.

Have anyone encountered this issue before? Any advice on how to solve this would be appreciated.  
================================================  
Screenshots:  

[image]  

[image]

liunao123, 2024-01-31:
hello，I have the same problem, can you tell me your solution? 3Q  -_-

supercrazysam, 2024-02-01:
> hello，I have the same problem, can you tell me your solution? 3Q -_-

I was not able to find a solution for this.  If you were able to crack this, please do let me know how.  
Much appreciated.  

Good luck on your debugging journey bro.  

Sincerely,  
Sam