Livox-SDK/Livox-SDK #216: livox sdk install at ROS2 rolling?

machine963, 2024-01-20:
when install 

after 

"cd build && cmake"

have error  on  "make" command

any 1 have same issus?

i mean install sdk , not driver

jdesbonnet, 2024-01-21:
I've been experimenting with the various repositories here over the past few days.  Some do take a some effort to build. I'm using docker to control the build environment as much as possible. What error are you getting?

F2M360, 2024-09-16:
Machine963
I'm having the same issues. right now. Were you able to fix it?

jdesbonnetfarmeye, 2024-09-16:
I took the approach of writing my own interface code according to the network protocol documentation. That's got me started with data acquisition.

F2M360, 2024-09-17:
I guess that's the issue I'm not computer savvy so it's not the same for me. I didn't know that I would have to program the lidar in order to use it. I tried to do it but my computer knowledge is limited. Can the program run on windows operating system? Would you be able to help me program it or have a more detailed steps on how to program the codes for ROS SDK? I need all the help that I can get.