Livox-SDK/Livox-SDK #220: Intalling SDK on Raspberry Pi 5 (ubuntu)

filippide, 2024-03-10:
Followind this steps:
sudo apt install cmake (OK)
git clone https://github.com/Livox-SDK/Livox-SDK.git (OK)
cd Livox-SDK (OK)
cd build && cmake .. 

c_______:~/Livox-SDK/build$ make
[  2%] Building CXX object sdk_core/CMakeFiles/livox_sdk_static.dir/src/base/io_loop.cpp.o
In file included from /home/cicci/Livox-SDK/sdk_core/include/third_party/spdlog/spdlog/fmt/fmt.h:21,
                 from /home/cicci/Livox-SDK/sdk_core/include/third_party/spdlog/spdlog/common.h:28,
                 from /home/cicci/Livox-SDK/sdk_core/include/../include/third_party/spdlog/spdlog/spdlog.h:12,
                 from /home/cicci/Livox-SDK/sdk_core/src/base/logging.h:27,
                 from /home/cicci/Livox-SDK/sdk_core/src/base/io_loop.cpp:30:
/home/cicci/Livox-SDK/sdk_core/include/third_party/spdlog/spdlog/fmt/bundled/format.h:407:6: error: identifier ‘char8_t’ is a keyword in C++20 [-Werror=c++20-compat]
  407 | enum char8_t: unsigned char {};
      |      ^~~~~~~
cmaecc1plus: note: unrecognized command-line option ‘-Wno-c++11-long-long’ may have been intended to silence earlier diagnostics
cc1plus: all warnings being treated as errors
make[2]: *** [sdk_core/CMakeFiles/livox_sdk_static.dir/build.make:90: sdk_core/CMakeFiles/livox_sdk_static.dir/src/base/io_loop.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:242: sdk_core/CMakeFiles/livox_sdk_static.dir/all] Error 2
make: *** [Makefile:136: all] Error 2

c______________:~$ sudo apt-get install gcc-aarch64-linux-gnu g++-aarch64-linux-gnu

[sudo] password for ______:
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Note, selecting 'gcc' instead of 'gcc-aarch64-linux-gnu'
Note, selecting 'g++' instead of 'g++-aarch64-linux-gnu'
gcc is already the newest version (4:13.2.0-1ubuntu1).
g++ is already the newest version (4:13.2.0-1ubuntu1).
0 upgraded, 0 newly installed, 0 to remove and 165 not upgraded.

My configuration is:

# System Details Report
---

## Report details
- **Date generated:**                              2024-03-10 00:11:11

## Hardware Information:
- **Hardware Model:**                              (null)
- **Memory:**                                      7,8 GiB
- **Processor:**                                   
- **Graphics:**                                    V3D 7.1
- **Disk Capacity:**                               (null)

## Software Information:
- **Firmware Version:**                            (null)
- **OS Name:**                                     Ubuntu 23.10
- **OS Build:**                                    (null)
- **OS Type:**                                     64-bit
- **GNOME Version:**                               45.0
- **Windowing System:**                            Wayland
- **Kernel Version:**                              Linux 6.5.0-1005-raspi
My configuration is:

18722013889, 2024-08-20:
raspberry pi 5 ubuntu 22.04 i got same issue，i have change gcc version from gcc 13.2.0 to gcc 9.5.0

Tiryoh, 2024-11-28:
I faced the same error, and this repo saved me a lot!

https://github.com/shenxw0127/Livox-SDK-24.04