Livox-SDK/Livox-SDK #222: PPS+GPRMC

leomessi999, 2024-03-26:
Currently, I am using GPS+PPS for radar time synchronization. I would like to ask what is the role of the PPS signal in the Horizon radar?

TokyoWarfare, 2025-11-14:
can you disconnect the pps signal from the livox converter 2.0 and see if still works?

If so, may be it would be possible not to use the livox converter 2.0 and instead use a cable like this:

https://es.aliexpress.com/item/1005010019740560.html?spm=a2g0o.productlist.main.5.1570HNuVHNuVWE&algo_pvid=39e04170-42cb-4fc5-b792-6342b5a94cf7&algo_exp_id=39e04170-42cb-4fc5-b792-6342b5a94cf7-4&pdp_ext_f=%7B%22order%22%3A%22-1%22%2C%22eval%22%3A%221%22%2C%22fromPage%22%3A%22search%22%7D&pdp_npi=6%40dis%21EUR%2129.14%2117.19%21%21%21234.61%21138.42%21%4021039a5b17630781258593613e8063%2112000050856566631%21sea%21ES%210%21ABX%211%210%21n_tag%3A-29910%3Bd%3Aa3128444%3Bm03_new_user%3A-29895&curPageLogUid=EwDXctB0V1Kf&utparam-url=scene%3Asearch%7Cquery_from%3A%7Cx_object_id%3A1005010019740560%7C_p_origin_prod%3A

Also if you could please share your GPS module and if you used a TTL to USB converter, tx.

TokyoWarfare, 2026-01-16:
I'm now at this stage of integrating the GPS. 

The role of the PPS is to reset to 0 the value of the second  of the internal clock. Keeping it as much in sync as possible with the clock of the satellite. Hence the timestamped coordinates will be as tightly coupled to the sensor's real world position as possible with whatever the current GPS accuracy is.  In short, it is a mean to reduce time drift.